Beckhoff/ADS #160: Is this library complementing the ADS libraries by Beckhoff?

jstiefel, 2022-03-07:
I'm trying to classify this library into the whole Beckhoff ecosystem. Beckhoff says that "ADS is the communication protocol of TwinCAT" and offers several API interfaces, e.g. DLL C++, .NET, DLL Java. Does this library provide another implementation of the same API with the same functionality? Why is it not officially listed then?

Is the API the same as the C++ API so that the same [examples listed here](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_ads_intro/index.html&id=) can be used?

We're developing a medical product and I'm trying to reduce third-party software as much as possible. So far I was just including the [AdsLib folder](https://github.com/Beckhoff/ADS/tree/master/AdsLib). I realized that you added two new folders `TwinCAT` and `standalone` there. Can you tell me more about the structure of the repository?

Is there a way to get support from Beckhoff for developing commercial products based on Linux?

mbollini, 2022-03-07:
Under Windows you use Beckhoff's DLL for ADS communication.  For example I use .NET wrapped in LabVIEW VIs to interface with the DLL.  In Linux or Mac OS for ADS communication you need to use this open-source library.

jstiefel, 2022-03-07:
> Under Windows you use Beckhoff's DLL for ADS communication.  For example I use .NET wrapped in LabVIEW VIs to interface with the DLL.  In Linux or Mac OS for ADS communication you need to use this open-source library.

Sorry, I probably wasn't very clear on this. I'm totally aware how to use these implementations. I'm more wondering how this implementation is connected to the others and who is maintaining it.

pbruenn, 2022-03-08:
@jstiefel As far as I know all the other Beckhoff libraries access a local TwinCAT ADS router to communicate, which means you need a TwinCAT basic installation on the host with your client application.
This library here _can_ do the same, in which case the implementation from `TwinCAT` is used. However, the main purpose of this OpenSource library is to run on systems _without_ TwinCAT router. In that case the code from `standalone` is used.  The new folders didn't introduce new dependencies! They just made the separation more clear!

So as far as I know on Linux your options are:                   
- use this OpenSource library                 
- find an alternative implementation (e.g. https://github.com/stamp/goADS)                                                           
- implement your own                                                                                                                                
                                                                                                                                                                                                          
Regarding, support from Beckhoff: Well, I help as much as I can with problems in this library, but from my experience most problems are more about specific ADS index groups and index offsets, which are properties of the ADS endpoint and not of this library. But as you most certainly communicate with some Beckhoff endpoint, of course our support will help you with that (you would need to contact them on the normal support channel, as they don't read here). And when it comes to our hardware, e.g. you want to run Linux on one of our industrial PC, we already wrote some drivers to better utilize our hardware on Linux, so you should be able to get support for that, too. You might ran into support people, who never heard about this library and Linux, and tell you they can't help you, this will not happen in bad faith! It's just that Beckhoff was a pure Windows company for multiple decades, and so are many of my colleagues. So please be gentle with them, probably they just don't know better.

jstiefel, 2022-03-08:
Thank you very much for your explanation about the background of this library, @pbruenn.

closed 2022-03-08 by jstiefel