Beckhoff/ADS #163: How to read all PLC Variables ?

vishnu-marthala, 2022-04-29:
Hello Everyone,

I couldn't find any documention or any function that gives me all the plc variables. How can show to the use in my UI all the avaiable PLC Symbols or Variables that are running on the controller? In Codesys i did it using GetAllItems() function which give all the variables along with their type and size. Is there similar function in ADS library as well? 

-  How can i access or read all avaiable PLC Symbols or variables that are running on the PLC Controller?
-  Which ADS libary should i use? There is Standalone and other one as well. Dont know the difference due to missing wiki
-  Is there any pdf or some documentation on how to use ADS library?

thank you!

Capacites, 2022-07-25:
https://github.com/Beckhoff/ADS/pull/170

pbruenn, 2023-01-13:
This is now available since commit 4cc5ded755f119d431f232502ffbe0688865f5ba

closed 2023-01-13 by pbruenn