Beckhoff/ADS #165: Suggestions on Integrating this Package with ROS

Dshieff, 2022-06-13:
Hi,

I would like to integrate this package into ROS and be able to call the functions in a ROS node, I would also like it to be able to build it using catkin. I have not adapted an existing package into ROS before and am not sure of the best way of adapting this package so I decided to ask here where there is a good chance someone has already done something like this before.

 Is the best approach to create a repository inside my catkin work space, then create a new CMakeLists.txt file using catkin, then copy + paste whatever is in the existing CMakeLists.txt files in the ADS library and then add my ROS specific makefile commands.

Is there anything in these CMakeLists.txt files that I need to be aware of that would mean that it would not work with ROS?

Or is there an easier/better way to do this?

AlexisFetet, 2022-07-22:
Hi, 

This library works well with ROS Melodic/Noetic (and probably more), and you can find [here](https://gitlab.univ-nantes.fr/CAPACITES-ROBOTIQUE_PROCEDES/module-ros/adsbeckhoff) an exemple of a ROS node.
We have tested it from a linux computer to both x86 and ARM automates and this was working just fine. However we encountered issues when working from an ARM computer with an ARM automate, that are currently being investigated.
The used ADS lib in this project has been slightly modified to add some more functionnalities (it is however a bit behind the current lib but we are working on having it up to date), but the existent function remain unchanged. A PR should be open in a near futur about this additions.

Dshieff, 2022-07-24:
Thank you for this! I'll have a look. I managed to modify the CMakeLists.txt files to enable it run as a ROS node but thank you for this solution as I'll use this in the future and it will give me some ideas on how to better write the modifications.

Capacites, 2022-07-25:
@Dshieff , we have publish a new version of the ROS ADS node.
We have used the default Cmakefile inside the project for Catkin_make.

Dshieff, 2022-07-25:
Ah thanks I'll have a look into that. I tend to use catkin build rather than catkin make so that might be where I had a few issues building the Beckhoff/ADS library.

closed 2023-01-11 by pbruenn