Beckhoff/ADS #176: Fail to compile on Windows 10

davidnhtd, 2022-10-20:
I am unable to compile ADS on Windows 10. These are the output of each command:
 
Meson build output:
```The Meson build system
Version: 0.63.3
Source dir: C:\ADS
Build dir: C:\ADS\build
Build type: native build
Project name: AdsLib
Project version: 0.1
Activating VS 17.3.6
C++ compiler for the host machine: cl (msvc 19.33.31630 "Microsoft (R) C/C++ Optimizing Compiler Version 19.33.31630 for x64")
C++ linker for the host machine: link link 14.33.31630.0
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library ws2_32 found: YES
Run-time dependency threads found: YES
Build targets in project: 4

Found ninja-1.11.1 at "C:\Program Files\Meson\ninja.EXE"

Visual Studio environment is needed to run Ninja. It is recommended to use Meson wrapper:
"C:\Program Files\Meson\meson.exe" compile -C build
```

ninja -C build output:
```
ninja: Entering directory `build'
[1/24] Compiling C++ object libAdsLib.a.p/AdsLib_AdsDef.cpp.obj
FAILED: libAdsLib.a.p/AdsLib_AdsDef.cpp.obj
"cl" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "/MD" "/nologo" "/showIncludes" "/utf-8" "/Zc:__cplusplus" "/W4" "/WX" "/EHsc" "/O2" "/Gw" "-D_FORTIFY_SOURCE=2" "/FdlibAdsLib.a.p\AdsLib_AdsDef.cpp.pdb" /FolibAdsLib.a.p/AdsLib_AdsDef.cpp.obj "/c" ../AdsLib/AdsDef.cpp
CreateProcess failed: The system cannot find the file specified.
[2/24] Compiling C++ object libAdsLib.a.p/AdsLib_AdsDevice.cpp.obj
FAILED: libAdsLib.a.p/AdsLib_AdsDevice.cpp.obj
"cl" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "/MD" "/nologo" "/showIncludes" "/utf-8" "/Zc:__cplusplus" "/W4" "/WX" "/EHsc" "/O2" "/Gw" "-D_FORTIFY_SOURCE=2" "/FdlibAdsLib.a.p\AdsLib_AdsDevice.cpp.pdb" /FolibAdsLib.a.p/AdsLib_AdsDevice.cpp.obj "/c" ../AdsLib/AdsDevice.cpp
CreateProcess failed: The system cannot find the file specified.
[3/24] Compiling C++ object libAdsLib.a.p/AdsLib_AdsLib.cpp.obj
FAILED: libAdsLib.a.p/AdsLib_AdsLib.cpp.obj
"cl" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "/MD" "/nologo" "/showIncludes" "/utf-8" "/Zc:__cplusplus" "/W4" "/WX" "/EHsc" "/O2" "/Gw" "-D_FORTIFY_SOURCE=2" "/FdlibAdsLib.a.p\AdsLib_AdsLib.cpp.pdb" /FolibAdsLib.a.p/AdsLib_AdsLib.cpp.obj "/c" ../AdsLib/AdsLib.cpp
CreateProcess failed: The system cannot find the file specified.
[4/24] Compiling C++ object libAdsLib.a.p/AdsLib_AdsFile.cpp.obj
FAILED: libAdsLib.a.p/AdsLib_AdsFile.cpp.obj
"cl" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "/MD" "/nologo" "/showIncludes" "/utf-8" "/Zc:__cplusplus" "/W4" "/WX" "/EHsc" "/O2" "/Gw" "-D_FORTIFY_SOURCE=2" "/FdlibAdsLib.a.p\AdsLib_AdsFile.cpp.pdb" /FolibAdsLib.a.p/AdsLib_AdsFile.cpp.obj "/c" ../AdsLib/AdsFile.cpp
CreateProcess failed: The system cannot find the file specified.
[5/24] Compiling C++ object libAdsLib.a.p/AdsLib_LicenseAccess.cpp.obj
FAILED: libAdsLib.a.p/AdsLib_LicenseAccess.cpp.obj
"cl" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "/MD" "/nologo" "/showIncludes" "/utf-8" "/Zc:__cplusplus" "/W4" "/WX" "/EHsc" "/O2" "/Gw" "-D_FORTIFY_SOURCE=2" "/FdlibAdsLib.a.p\AdsLib_LicenseAccess.cpp.pdb" /FolibAdsLib.a.p/AdsLib_LicenseAccess.cpp.obj "/c" ../AdsLib/LicenseAccess.cpp
CreateProcess failed: The system cannot find the file specified.
[6/24] Compiling C++ object libAdsLib.a.p/AdsLib_Log.cpp.obj
FAILED: libAdsLib.a.p/AdsLib_Log.cpp.obj
"cl" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "/MD" "/nologo" "/showIncludes" "/utf-8" "/Zc:__cplusplus" "/W4" "/WX" "/EHsc" "/O2" "/Gw" "-D_FORTIFY_SOURCE=2" "/FdlibAdsLib.a.p\AdsLib_Log.cpp.pdb" /FolibAdsLib.a.p/AdsLib_Log.cpp.obj "/c" ../AdsLib/Log.cpp
CreateProcess failed: The system cannot find the file specified.
[7/24] Compiling C++ object libAdsLib.a.p/AdsLib_RouterAccess.cpp.obj
FAILED: libAdsLib.a.p/AdsLib_RouterAccess.cpp.obj
"cl" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "/MD" "/nologo" "/showIncludes" "/utf-8" "/Zc:__cplusplus" "/W4" "/WX" "/EHsc" "/O2" "/Gw" "-D_FORTIFY_SOURCE=2" "/FdlibAdsLib.a.p\AdsLib_RouterAccess.cpp.pdb" /FolibAdsLib.a.p/AdsLib_RouterAccess.cpp.obj "/c" ../AdsLib/RouterAccess.cpp
CreateProcess failed: The system cannot find the file specified.
[8/24] Compiling C++ object libAdsLib.a.p/AdsLib_RTimeAccess.cpp.obj
FAILED: libAdsLib.a.p/AdsLib_RTimeAccess.cpp.obj
"cl" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "/MD" "/nologo" "/showIncludes" "/utf-8" "/Zc:__cplusplus" "/W4" "/WX" "/EHsc" "/O2" "/Gw" "-D_FORTIFY_SOURCE=2" "/FdlibAdsLib.a.p\AdsLib_RTimeAccess.cpp.pdb" /FolibAdsLib.a.p/AdsLib_RTimeAccess.cpp.obj "/c" ../AdsLib/RTimeAccess.cpp
CreateProcess failed: The system cannot find the file specified.
[9/24] Compiling C++ object libAdsLib.a.p/AdsLib_Sockets.cpp.obj
FAILED: libAdsLib.a.p/AdsLib_Sockets.cpp.obj
"cl" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "/MD" "/nologo" "/showIncludes" "/utf-8" "/Zc:__cplusplus" "/W4" "/WX" "/EHsc" "/O2" "/Gw" "-D_FORTIFY_SOURCE=2" "/FdlibAdsLib.a.p\AdsLib_Sockets.cpp.pdb" /FolibAdsLib.a.p/AdsLib_Sockets.cpp.obj "/c" ../AdsLib/Sockets.cpp
CreateProcess failed: The system cannot find the file specified.
[10/24] Compiling C++ object libAdsLib.a.p/AdsLib_Frame.cpp.obj
FAILED: libAdsLib.a.p/AdsLib_Frame.cpp.obj
"cl" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "/MD" "/nologo" "/showIncludes" "/utf-8" "/Zc:__cplusplus" "/W4" "/WX" "/EHsc" "/O2" "/Gw" "-D_FORTIFY_SOURCE=2" "/FdlibAdsLib.a.p\AdsLib_Frame.cpp.pdb" /FolibAdsLib.a.p/AdsLib_Frame.cpp.obj "/c" ../AdsLib/Frame.cpp
CreateProcess failed: The system cannot find the file specified.
ninja: build stopped: subcommand failed.
```

Does anyone have the same problem? Thanks

pbruenn, 2022-10-24:
Did  you ran both commands from a "Visual Studio Developer Command Prompt"?

davidnhtd, 2022-10-24:
I was running on normal command prompt. After reading your comment, I tried to run the commands with VS developer command prompt and could see more details of why it fails. There are 2 main problems:
- the useage of std::numeric_limits::max and min in Windows causes build errors. I followed a temp solution of #undef as explained [here](https://stackoverflow.com/questions/27442885/syntax-error-with-stdnumeric-limitsmax). I am not sure what the correct solution would be.
- some other build warnings being treated as error because -werror flag is set. For this, I temporarily set the flag to false and the build doesnot fail on warnings anymore

After fixing these, I could successfully build.

pbruenn, 2022-10-24:
You find the full workaround I use here: https://github.com/Beckhoff/ADS/blob/master/.gitlab-ci.yml#L81. With that you should be able to compile warning free (until you use TwinCAT header files).

closed 2023-01-11 by pbruenn