Beckhoff/ADS #177: Error 6 when running on Windows 10 MinGW build

dvlretiina, 2022-11-23:
## Problem
Build the project with MinGW (MSYS2 UCRT) on Windows 10 and add routes and change to correct remote/local AMSNetID, and IPV4 and it fails with Error 6.

Building the exact same code in WSL works, connects and reads/writes ADS vars just fine.

## Issue
It appears that the `InitSocketLibrary` function is NOT called before `getaddrinfo` in `GetListOfAddresses` which is ultimately called from the `AdsDevice` constructor.

## Workaround
For now it seems that adding my own call to `InitSocketLibrary` before calling the `AdsDevice` constructor (or at the beginning of main before `runExample`) solves the problem.

## Notes
Additionally, the `ws2_32` library, although linked as PUBLIC is not automatically included in external projects that link to the ADS library and I've had to add it myself.

Please inform me if I'm missing some documentation that references this or if I've made invalid assumptions.

pbruenn, 2022-11-23:
Either adjust you build script to pass _WIN32 or __CYGWIN__ or tell us how your platform can be detected and we can adjust:
[#if !(defined(_WIN32) && !defined(__CYGWIN__))](https://github.com/Beckhoff/ADS/blob/a1e04813d74b55cf6abcaa25982eb89b5dd811e7/AdsLib/wrap_socket.h#L10)

Your linker issue sounds like CMake specific? I have no knowledge with that, maybe someone from the community can improve the CMakeList.txt files.

pbruenn, 2022-11-23:
Oh wait, you say you call InitSocketLibrary()??? Not WSAStartup?

dvlretiina, 2022-11-23:
I'm running on Windows 10 within the MSYS UCRT environment with MinGW (latest download as of yesterday https://github.com/msys2/msys2-installer/releases/download/2022-10-28/msys2-x86_64-20221028.exe).

That is correct I mean `InitSocketLibrary` and not `WSAStartup` call directly.  I see that `InitSocketLibrary` appropriately calls `WSAStatrup`, and is build correctly for _WIN32 platforms.  HOWEVER, I believe `InitSocketLibrary` is not actually being called before the first call to `getaddrinfo` in `GetListOfAddresses`.  In fact, I can see this is the case.  This appears to be why, if I add a call to `InitSocketLibrary` within the example, it runs successfully and no longer reports Error 6.

The example project will build and correctly detects (within CMake build) the WIN32 and MINGW settings.  My project that uses ADS also builds correctly and detects the same CMake vars, but does not appear to benefit from the inclusion of `ws2_32` library from within ADS.  I have to manually add it myself to my own project (when it detects WIN32 platform in CMake), otherwise I see the following linker error:

```
FAILED: src/test_adsinterface.exe 
cmd.exe /C "cd . && E:\msys64\ucrt64\bin\c++.exe -O3 -DNDEBUG  src/CMakeFiles/test_adsinterface.dir/test_adsinterface.cpp.obj -o src\test_adsinterface.exe -Wl,--out-implib,src\libtest_adsinterface.dll.a -Wl,--major-image-version,0,--minor-image-version,0 -LE:/msys64/usr/local/lib -lads  fmt/libfmt.a  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
E:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/CMakeFiles/test_adsinterface.dir/test_adsinterface.cpp.obj:test_adsinterface.cpp:(.text+0x14c): undefined reference to `__imp_WSAStartup'
collect2.exe: error: ld returned 1 exit status
```

closed 2022-11-25 by pbruenn