Beckhoff/ADS #185: TwinCAT ADS  Error code 1873(Hash table overflow)..

AnandHegden, 2023-01-31:
Hi, Please could I know the reason for getting error code 1873 while registering the variable in TwinCAT ADS Communication and what is the possible way to solve the issue?

pbruenn, 2023-01-31:
Without knowing more I would guess that you are registering too many variables

AnandHegden, 2023-01-31:
> Without knowing more I would guess that you are registering too many variables

added all reuired register only..but how to resolve it  or any limitation on registering variable?

pbruenn, 2023-01-31:
> > Without knowing more I would guess that you are registering too many variables
> 
> added all required register only..but how to resolve it or any limitation on registering variable?

Do you consider this more info? Sorry, I can't help you and I guess nobody else can, if you don't start telling us what you are doing.

AnandHegden, 2023-01-31:
> 

Please could I know what kind of information need?

AnandHegden, 2023-02-01:
Please could I know, what kind of info need?

AnandHegden, 2023-02-01:
Hi, I am using TwinCAT 3.0 ADS Communication using for  C++ MFC Application and used TcAdsDll API while registering the variable getting ADS comm error code 1873(Hash table overflow error)..Could you please know the reason and all possible ways to solve the issue?

chrisbeardy, 2023-10-12:
@pbruenn 

I have also started getting this issue, this happens when getting some variable handles. Does the Error code 1873 mean you are trying to acquire too many handles in one go, or just too many over the course of twincat running? 

I have an application which is running 24/7, every say 60 seconds is will acquire handles, read the variables for 10 seconds, then release the handles. I only get this error every now and again which is odd as I would assume that as I am releasing the handles it should be ok.  How does ADS respond if you try to acquire the handle to a variable you already have acquired a handle for, doe it creates another "reference" / hash table entry, i.e. could it be I am sometimes not releasing the handles?

How would I recover from this error? the only way I have found so far is rebooting the PLC.

pbruenn, 2023-10-14:
Sorry, my knowledge ends when the TCP packet is send from the socket. The error is emitted by the ADS device on the other side so the meaning of 1873 depends on the specific ADS device. If it is a TwinCAT 2 or TwinCAT 3 PLC, please ask your Beckhoff support about the meaning. This is a generic problem and not related to this open source library, so they should be able to help you.

chrisbeardy, 2023-12-04:
I have been doing some experimentation, and found the cause of the issue. 

For your interest:

- If you take out a variable handle, then take out the same handle again and again without releasing it, you get the same handle id back and this does not appear to affect any hash table.
  - I tested this by taking out the same variable handle on loop for a number of hours.
- If you create a notification handle,  then take out the same handle again and again without releasing it, you get different handle ids each time, your available router memory then starts to lower and you very quickly get the 1873 hash table overflow error.

In my case, there was some very specific behaviour in the programming language (garbage collection in python) which meant that my notification handles were not being deleted as I expected. This means I have now solved my problem by making sure I actively delete my handle.

pbruenn, 2023-12-04:
@chrisbeardy thanks for reporting back!

closed 2023-12-04 by pbruenn