Beckhoff/ADS #186: ADS error 0x701 (1793) when reading SystemId

mtheikkila, 2023-01-31:
Not possible to read SystemId from the Beckhoff IPC, PLC in run state.

1. Background:
Cloned the repository, set PLC up and running in separate IPC  and start testing example code in Debian 11.
https://github.com/Beckhoff/ADS/tree/master/example
updated IP addresses and AmsNetId -> build and tested communication with the PLC. -> Works (all the variables are read/wrote successfully) :)

2. Added lines of code to read the system id
example code:
auto device = bhf::ads::LicenseAccess{ std::string(remoteIpV4), remoteNetId, AMSPORT_R0_PLC_TC3 };
device.ShowSystemId(std::cout);
device.ShowPlatformId(std::cout);
device.ShowVolumeNo(std::cout);

Result:
2023-01-31T10:13:42+0200 Error: ShowSystemId(): failed with: 0x701
2023-01-31T10:13:42+0200 Error: ShowPlatformId(): failed with: 0x701
2023-01-31T10:13:42+0200 Error: ShowVolumeNo(): failed with: 0x701

3. adstool gives:
./adstool 5.55.129.238.1.1 license systemid
2023-01-31T10:30:33+0200 Error: AdsException message: Ads operation failed with error code 6.

pbruenn, 2023-01-31:
The SystemId is not supported by AMSPORT_R0_PLC_TC3. You have to ask the R0_LICENSESERVER [1].  
So either do:  
`auto device = bhf::ads::LicenseAccess{ std::string(remoteIpV4), remoteNetId, 30};`  
or just let LicenseAccess use its default by setting the port to 0:  
`auto device = bhf::ads::LicenseAccess{ std::string(remoteIpV4), remoteNetId, 0};`    

[1] https://infosys.beckhoff.com/content/1033/tc3_ads.net/9408352011.html

mtheikkila, 2023-01-31:
of course! Problem solved. Thanks for help.

closed 2023-01-31 by mtheikkila