Beckhoff/ADS #188: AdsSyncReadReqEx2() returns -1

ArsMasiuk, 2023-02-17:
On Ubuntu, AdsSyncReadReqEx2 returns code "-1" which is not listed anywhere.

What does it mean and how to resolve the issue?

The same call under Windows works pretty fine,

pbruenn, 2023-02-18:
I guess it is this one:
https://github.com/Beckhoff/ADS/blob/master/AdsLib/standalone/AmsConnection.cpp#L148

Which means [AmsConnection::Write()](https://github.com/Beckhoff/ADS/blob/master/AdsLib/standalone/AmsConnection.cpp#L105) returned a nullptr. If you don't see any other log output this means not all of your data was written to the [socket](https://github.com/Beckhoff/ADS/blob/master/AdsLib/standalone/AmsConnection.cpp#L126-L129).

ArsMasiuk, 2023-02-20:
Yes, it look like so. But what could be the reason? 
There is a "normal" 1-GB network card, with Jumbo Frames enabled (MTU=9000).
Under Windows, it works with no errors. Am I missing something?

ArsMasiuk, 2023-02-20:
Hmm, I dont see these error messages, but there come warnings like:

Warning: Port: 30000 already in use as <and some hex value here>

So it looks like it is getting failed even before writing to the socket,  in AmsConnection::Reserve() at line 119 in AmsConnection.cpp.

I see port 30000 when the app is getting connected to the PLC.
The application is multithreaded, so it opens the port in one thread but does further communication in another one.
Is it not allowed on Linux?

ArsMasiuk, 2023-02-20:
Ah ok I see... 
So the application has to be changed in order to access AMS via only one thread in order to get it working on Linux.
Thank you Patrick for your help!

closed 2023-02-20 by ArsMasiuk