Beckhoff/ADS #190: Run ADS Client on WSL through localhost

captain-yoshi, 2023-02-22:
Is it possible to run the ADS Client on a _windows subsystem linux_ (WSL) which communicates through the localhost ? FYI it works with another PC using the same linux distribution as the WSL (Ubuntu 20.04).

Maybe related to #93 and #141.

**ADS Route**

[image]

**Client Config**
```cpp
static void runExample(std::ostream& out)
{
    static const AmsNetId remoteNetId { 192, 168, 0, 193, 1, 1 };
    static const char remoteIpV4[] = "192.168.0.193";

    // uncomment and adjust if automatic AmsNetId deduction is not working as expected
    // Tried with and without this line below
    // bhf::ads::SetLocalAddress({192, 168, 0, 193, 1, 1});
    ...
}
```

**Running the Client Example**
```bash
~$ ./example/build/example
# Nothing for 5 seconds and then :
2023-02-22T15:04:20-0500 Info: connection closed by remote
Error: 1861
AdsException message: Ads operation failed with error code 1861.
```
---

**Wireshark Debug (loopback adapter)**
[wireshark-capture.zip](https://github.com/Beckhoff/ADS/files/10807867/wireshark-capture.zip)

[image]

pbruenn, 2023-02-28:
TwinCAT doesn't expect TCP connections from localhost. Especially, when you use the same AMS NetId as Target and Sender address. So at least you have to use different AMS NetIds for the TwinCAT server (in outside Windows) and your ADS Client (inside WSL). Then it depends on WSL networking, which I have no clue about. If on the TwiNCAT side the connections looks like a normal remote host, then it should work as soon as you configured the AMS route correctly. If TwinCAT receives a TCP connection attempt from an IP it is listening itself I doubt it will work.

captain-yoshi, 2023-03-01:
No luck with different AMS NetId... WSL1 gives you a localhost network only. WSL2 gives you a virtual network that you can configure but needs Hyper-V which is not compatible with TwinCAT in run-mode.

Thanks for the library ! Very handy 👍

klauer, 2023-10-13:
I think the author of a nodejs ADS client figured out what's necessary for localhost communication - reservation of a port on the router. This has since been at least partially propagated to the rust client ads-rs (though I'm not sure if it cleared the reserved port after use). It should work for WSL1 with localhost networking, though WSL2 is a different story as you've noted.

Since this is technically possible over ADS, I wonder if it's something that might be accepted for Beckhoff's open source library here. Even some official acknowledgment/documentation of the underlying mechanism would be nice, too.

Links:
* nodejs reference: [registration](https://github.com/jisotalo/ads-server/blob/c9d09d6af4e4137aa555b4bd9ba2c18d4ce55a81/src/ads-server-router.ts#L484-L575), [unregistration](https://github.com/jisotalo/ads-server/blob/c9d09d6af4e4137aa555b4bd9ba2c18d4ce55a81/src/ads-server-router.ts#L580-L650)
* Rust reference: https://github.com/birkenfeld/ads-rs/blob/7d4ca8a81f86e2fac14432c39df9ec20d352ab99/src/client.rs#L219-L230
* My WIP Python implementation of this: https://github.com/pcdshub/ads-async/pull/40
* Related issues: #19 #93