Beckhoff/ADS #193: ADS TwinCAT3 And Linux Virtual Machine Communication : Error: 1861

ismailalhajosman, 2023-03-09:
Hello, 

I have my TwinCAT3 running on Winodws 10 and want to communicate with Linux OS running on Oracle Virtual Machine. I followed all the steps written in the [readme](https://github.com/Beckhoff/ADS#readme) and run example.cpp file to do the communication however I get this error. I followed all issues related to this problem including #14 and #56  but could not solve this error.

This is my part of the code to assign PLC  remote IP :
[image]

And this my PLC IP and Router settings : 
[image]

Any help ? 

Thanks in advance !

pbruenn, 2023-03-09:
I assume N_V_M is you Linux virtual machine and the Static Routes screenshot is from the Windows host?

In that case the AdsSetLocalAddress is wrong.  Your route is configured for 192.168.108.212.1.1, but your AdsSetLocalAddress() uses 192.168.0.114.1.1

ismailalhajosman, 2023-03-10:
Yes that is correct.

The problem is solved my changing  my AdsSetLocalAddress() to 192.168.108.212.1.1 

Thanks a lot!

closed 2023-03-10 by ismailalhajosman