Beckhoff/ADS #200: Ping a AMS  NetId

DonMiguelSan, 2023-05-11:
I have a function to parse the xml file with the static routes, nevertheless I want to ping them to ensure which of them are available and with the ads port opened (in Run mode), at the moment I perform a connect operation to check the ads status, but it seems too complicated to just test if they are available and connected. is there an official supported way to do this?

pbruenn, 2023-05-11:
When I try to "ping" an ADS service I query its state e.g. like this https://github.com/Beckhoff/ADS/blob/master/example/example.cpp#L104

or in plain ADS:
https://infosys.beckhoff.com/content/1033/tc3_adsdll2/117543691.html?id=4954386139193486380

soberschmidt, 2023-05-12:
You can read the adsStates from every amsNetId in staticroutes.xml from the system service (amsPort=10000) @pbruenn already mentioned.

DonMiguelSan, 2023-05-12:
Great info, i have to implement it also in c#, i did not realized that i was in cpp Repository, is there something similar for c#? i just looked in the dll but there are only two enum availables in that namespace, any suggestions? 
[image]

soberschmidt, 2023-05-12:
Check [https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads]

DonMiguelSan, 2023-05-12:
Thank you very much that is exactly what i was looking for

DonMiguelSan, 2023-05-16:
Hello again,

i have checked the suggested information but unfortunately ended being not exactly what i was looking for, this provided solution was ok to read the status of the local AMSNETID, nevertheless it does not provide the capability to show the status of every static route configured in the machine:

[image]

My aim is to read the route and check the state with a single call, at the moment i implemented a "Broadcast" method to achieve this, but for me it is not the optimal way to do it.

At the moment i read the static routes xml file and perform with every result of the list a connect , after that i read the ads status with the same method provided by you guys. 

Is there something similar supported or do i have to connect first to every ads route to check the status?

Downside from this approach: If the ip or the AMSNETID do not exist anymore that i have to wait a timeout from the connect method, and as soon as you have 5 static routes it is too much time lost

pbruenn, 2023-05-23:
Well, this is a typical network issue. There is no magic I know of to fix this problem.

You can run the checks in parallel to reduce the overall amount of time. And you can reduce the timeout to wait even less.

closed 2023-08-04 by DonMiguelSan