Beckhoff/ADS #201: Two application running on the same machine that read a variable from the same plc concurrently

AlessandroMenon, 2023-05-11:
Hi, we have the following scenario:

1. First application that use [ads](https://github.com/Beckhoff/ADS) to read a variable from plc
2. Second application that use https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads to read the same variable from the same plc

Is this a supported scenario ? The behavior we see is that as soon as the second application starts, the first application is closed connection with the plc (" connection closed by remote"). If we start the two applications in reverse order we have the same problem on the other instance. So it seems that they cannot coexist on the same machine.
I attached the example
[Example.zip](https://github.com/Beckhoff/ADS/files/11452895/Example.zip)

AlessandroMenon, 2023-05-15:
Thank you for your answer. But I don't understand why if I use two instances of the .NET application I don't encounter this problem. It only happens to me if either application uses https://github.com/Beckhoff/ADS. Could it be because the .NET version has a real Ads Router implemented?

pbruenn, 2023-05-15:
Because the .NET application is "cheating" and implements an ADS router ;-)

AlessandroMenon, 2023-05-15:
Is there any way to use an ADS router in C++? Do I have to compile https://github.com/Beckhoff/ADS with the define USE_TWINCAT_ROUTER set and have TwinCAT installed on the machine?

pbruenn, 2023-05-15:
Yes, on Windows or TC/BSD you can build with `-Dtcadsdll_include=c:\TwinCAT\AdsApi\TcAdsDll\Include -Dtcadsdll_lib=c:\TwinCAT\AdsApi\TcAdsDll\x64\lib`, which will then use the TwinCAT Router.

Take a look into our internal CI scripts:
TC/BSD:
https://github.com/Beckhoff/ADS/blob/master/.gitlab-ci.yml#L68-L74

Windows (Is essentially the same but looks so complicated because of our CI setup):
https://github.com/Beckhoff/ADS/blob/master/.gitlab-ci.yml#L101-L105

fengyc, 2023-06-14:
Try this https://github.com/fengyc/ads-proxy , I don't think it is an elegant solution but it works

closed 2023-09-04 by pbruenn

bsteenput, 2023-12-06:
Hello, 

Is it possible to have the ads router in linux ? I'm trying to have the same machine to access different tasks running in the PLC the first process on the linux machine gets disconnected when the second process connect. I'm using PyAds on a ubuntu machine. 

My problem is the same as the one describe here : https://github.com/Beckhoff/ADS/issues/201

Thanks in advance for you help,
Bertrand

tboegi, 2023-12-06:
Did you try the suggestion from https://github.com/stlehmann/pyads/issues/331 ?

Otherwise:
https://github.com/birkenfeld/ads_forwarder_rs