Beckhoff/ADS #213: SymbolAccess for Bool

simonschmeisser, 2023-09-05:
If I use `SymbolAccess` to write a `BOOL` variable, the TwinCat Life view turns to "<Value of the expression can not be retrieved>". If I read it then it will be sometimes values like 48 (false) or 49 (true) instead of 0 and 1 as I would expect.

I write by using `symbols.Write("MAIN.boolVar", "1");` and read using `(int)readBuffer.data()[0]` (copy pasted from `SymbolAccess`)

tboegi, 2023-09-05:
"1" is a string, isn't it ?
`symbols.Write("MAIN.boolVar", 1);`

simonschmeisser, 2023-09-05:
No, there is no such function signature ...

pbruenn, 2023-09-06:
> "1" is a string, isn't it ? Does this work:
> 
> `symbols.Write("MAIN.boolVar", 1);`

Yeah, thats a bug because we handle uint8_t just like the other integer types but stringstream interprets it as ascii. I have a fix ready will release soon together with the string handling issue #194

simonschmeisser, 2023-09-06:
Good catch! I switched to using `AdsVariable` for those types but I wonder if there should not be more overloads like `Write(string, bool)` to avoid the string conversion all together?

Is there actually a technical difference between those two APIs? Do they result in different data being send to the PLC?

pbruenn, 2023-09-07:
Yes, there is a huge difference AdsVariable with handle is much safer! Accessing PLC variables "directly" by index group and offset is pretty dangerous, because you can silently corrupt the memory when the PLC did an online change for example. SymbolAccess was meant to demonstrate how access PLC variables in a generic way. A better version of SymbolAccess::Read/Write would generate a handle and use the handle when accessing the variable on the PLC.

However, the bug about ASCII instead of integer for BOOL values should be fixed with https://github.com/Beckhoff/ADS/commit/d7ada1228c68a7c4b2b86d357720ed8f6e76662c

closed 2023-09-07 by pbruenn