Beckhoff/ADS #215: Async WriteReqEx

captain-yoshi, 2023-09-13:
I am controlling a robot and need to exchange data at 1kHz. For receiving data, we use the AdsNotification which works well. For sending the data, we use the `WriteReqEx` function which is synchronuous, e.x. it will send the data and wait for the response. This process can take up from 1 to 2 msec (roundtrip). I don't need to wait for the response (we use a watchdog).

What I want is the ability to send the data without waiting for the response (similar to the AdsNotification). Is there an async `WriteReqExsuch` in the current API ? Could you give pointers on how to achieve this otherwise ?

```cpp
error = route_->WriteReqEx(ADSIGRP_SYM_VALBYHND,      // IndexGroup
                           *(handle_command_->get()), // IndexOffset
                           write_len,                 // BufferLength
                           (void *)&buffer_command_); // WriteBuffer

```

pbruenn, 2023-09-13:
ADS runs over TCP so fire & forget like you could do with UDP is not possible. At least your TCP send would wait for the ACK.  
If you don't care about the responses, you can either try `SetTimeout(0)` and/or build a queue, setup a pool of worker threads and let them send out the request.

captain-yoshi, 2023-09-14:
When using a pool of worker threads I get this :
```bash
[ERROR] [1694658053.675123296]: AdsException message: Ads operation failed with error code -1.
2023-09-13T22:20:53-0400 Warning: Port: 30000 already in use as 0x7fb3c5ffaba0
```
But it seems to work... On linux, I increment a watchdog and log this variable in twincat. When analyzing the data, I only miss 6 samples for the duration of the logging (4002 cycles).

Before, I missed 50 samples for the same amount of cycles.

captain-yoshi, 2023-09-16:
Great! No more samples missed with 10 worker threads.

Thanks again for pointing me to the right direction :)

closed 2023-09-19 by pbruenn