Beckhoff/ADS #216: AdsNotification cycle time deviation

captain-yoshi, 2023-09-26:
This is similar to #215. I need to get a reliable stream of data (304 Bytes) at 1kHz using the `AdsNotification` approach. I get inconsistent timing (see timing below). This can be due to:
- TCP
- The task settings on TwinCAT side
- The number of CPU ressources available at the receiving end (callback, OS, I/O's)
- Other reasons...

For the benchmark below, I have reduce my Twincat program to be as fast as possible and the callback on linux does nothing.

Is there a way to have a better timing tolerance between each packets received. Could the server side use a pool of threads or something else entirely to improve this behavior ? 

[image]

```cpp
route_notification_ = std::make_shared<AdsDevice>(
    remoteIpV4_, remoteNetId_, AMSPORT_R0_PLC_TC3);

// Get handle for command and state variables
handle_command_ = std::make_unique<AdsHandle>(
    ads_devices_[0]->GetHandle(adsBufferCommandName_));

// create notification callback for state
Callback<void(const AmsAddr *pAddr,
              const AdsNotificationHeader *pNotification,
              uint32_t hUser)>::func =
    std::bind(&RosTwincatBridge::NotifyCallback, this,
              std::placeholders::_1, std::placeholders::_2,
              std::placeholders::_3);
callback_t func = static_cast<callback_t>(
    Callback<void(const AmsAddr *pAddr,
                  const AdsNotificationHeader *pNotification,
                  uint32_t hUser)>::callback);

// Each server cycle @ 1kHz
AdsNotificationAttrib attrib = {
    DATA_STATE_SIZE, ADSTRANS_SERVERCYCLE, 0, {10000}};

notification_state_ = std::make_unique<AdsNotification>(
    *route_notification_, adsBufferStateName_, attrib, func, 2);

```

```sh
# Elapsed time between previous and current callback (1kHz -> 0.001 seconds)
AdNotification= 0.00117044 seconds
AdNotification= 0.00093865 seconds
AdNotification= 0.000897969 seconds
AdNotification= 0.0010241 seconds
AdNotification= 0.000986944 seconds
AdNotification= 0.00182602 seconds
AdNotification= 0.000157791 seconds
AdNotification= 0.00118531 seconds
AdNotification= 0.000804247 seconds
AdNotification= 0.000995989 seconds
AdNotification= 0.00100133 seconds
AdNotification= 0.00100328 seconds
AdNotification= 0.00103729 seconds
AdNotification= 0.00104816 seconds
AdNotification= 0.000939896 seconds
```

pbruenn, 2023-09-27:
Your first bullet point is the culprit. Remove TwinCAT and ADS from your benchmark and run it on a bare TCP connection. You will see most of your Jitter is systematic to TCP/IP.

soberschmidt, 2023-10-02:
ADS is not a fieldbus and was designed for acyclic communication. ADS notifications provide the option to collect data from _every_ PLC cycle and you will not miss a value (cmp. to polling) of your 1Khz signal. But TCP/IP (even if UDP is used) as already mentioned and the OS (UM/RT context switching) will cause a jitter, even if your ADS client and PLC runs on the same system.

For deterministic TCP communication with TwinCAT systems you can use network variables (TwinCAT 2), [EAP ](https://infosys.beckhoff.com/content/1033/eap/1521519371.html?id=795366871337330973)(TwinCAT 3) or [TwinCAT 3 TCP/UDP RT function](https://infosys.beckhoff.com/content/1033/tf6311_tc3_tcpudp/1076840971.html?id=8616395486301233937).

captain-yoshi, 2024-11-05:
Thanks ! I have a Linux PC on one side and a PC with Windows running a TwinCAT program on the other side. 

I'm also reading about [EtherCAT Master](https://www.acontis.com/en/ethercat-master-linux.html) software stacks, which uses cyclic data exchange via PDO mapping. I think I would need a real time driver + real-time kernel to match EAP deterministic communication... 

This [acontis blog](https://www.acontis.com/en/ethercat-master-options-ec-master-vs-open-source-etherlab-SOEM.html)  reviews 3 stacks (check the Supported EtherCAT Master Features section):
- Acontis EC-Master
- IgH EtherCAT-Master
- [SOEM library](https://github.com/OpenEtherCATsociety/SOEM) 

I will probably test out SOEM with a Linux PREEMPTED kernel to see if I can have more reliable data exchanges.

----
Can EAP be implemented on Linux ?

pbruenn, 2024-11-05:
> 
> Can EAP be implemented on Linux ?

Hopefully, it should be supported by https://www.beckhoff.com/en-en/company/news/multimedia-twincat-runtime-for-linux-r-from-beckhoff.html