Beckhoff/ADS #225: Info: connection closed by remote in sockets.cpp

lbhopper, 2024-03-14:
Hello,

No issue, just looking for some assistance.  I have modified the example.cpp to read my PLC and it works well. The only thing I cannot figure out is how to suppress the 'Info: connection closed by remote' when the script is finished. I've tried modifying the sockets.cpp file (lines 158 to 176, removing the (lastError == CONNECTION_CLOSED)), but have not had any luck.

Would someone be able to assist?

```
size_t Socket::read(uint8_t* buffer, size_t maxBytes, timeval* timeout) const
{
    if (!Select(timeout)) {
        return 0;
    }

    const auto msvcMaxBytes = static_cast<int>(std::min<size_t>(std::numeric_limits<int>::max(), maxBytes));
    const int bytesRead = recv(m_Socket, reinterpret_cast<char*>(buffer), msvcMaxBytes, 0);
    if (bytesRead > 0) {
        return bytesRead;
    }
    const auto lastError = WSAGetLastError();
    if ((0 == bytesRead) || (lastError == CONNECTION_ABORTED)) {
        throw std::runtime_error("connection closed by remote");
    } else {
        LOG_ERROR("read frame failed with error: " << std::dec << std::strerror(lastError));
    }
    return 0;
}
```

pbruenn, 2024-03-15:
seem you hit `bytesRead == 0`. so you can change the conditional above to `if (bytesRead >= 0) {`

lbhopper, 2024-03-15:
ok thank you

closed 2024-03-18 by lbhopper