Beckhoff/ADS #226: twincat3 - ubuntu error code 6

kkk3449, 2024-04-03:
Our current goal is to establish TCP/IP communication through an Ethernet cable between an Intel NUC (Ubuntu 20.04) - BECKHOFF PC CX5140 (TwinCAT3). The most desired method of communication is ROS-TwinCAT3.

For communication, we have:
Successfully made the ADS git on the Ubuntu PC,

Communicated using ADSLib based on C++ for ROS,

Communicated using pyads,

Communicated using C++ based ADSLib from the ADS GitHub,

Added a route connection on the BECKHOFF PC.

However, all attempts have failed due to the same error code 6. We have disabled the firewalls on both the Ubuntu and BECKHOFF PCs.

We have set up the following in this order.

1. BECKHOFF PC IP, AmsNetId setting
- IP : 192.168.0.2
- AmsNetId: 192.168.0.2.1.1
- config mode
    
    [image]
    

[image]

- Add Route

[image]

- Also add at StaticRouts.xml( 다른 이슈 글을 보고 이렇게 세팅했습니다)

[image]

2. ubuntu setting

ubuntu : 192.168.0.5

AmsnetId: 192.168.0.5.1.1

[image]

- check Routs

[image]

- try method 1 : example.cpp

[image]

- ADSError: error code 6

[image]

- try method 2 :pyads

```python
import pyads

plc = pyads.Connection('192.168.0.2.1.1', 851)  

value = plc.read_by_name('MAIN.myFloatVAR')  

plc.write_by_name('MAIN.myFloatVAR', 123, pyads.PLCTYPE_INT)  
plc.close()
```

- error code 6 : ADSError: target port not found   ADS Server not started (6).

[image]

- try method3 : ros (repo: [[twincat_talker](https://github.com/lemanhtrung/twincat_talker)](https://github.com/lemanhtrung/twincat_talker))

[image]

```cpp
#include "AdsLib.h"
#include "AdsNotification.h"
#include "AdsVariable.h"
#include "ros/ros.h"

int main(int argc, char *argv[]) {
  // ROS node init
  ros::init(argc, argv, "tctalker");
  ros::NodeHandle n;

  ROS_INFO_STREAM("running good");

  static const AmsNetId remoteNetId{192, 168, 0, 2, 1, 1};
  static const char remoteIpV4[] = "192.168.0.2";

  AdsDevice route{remoteIpV4, remoteNetId, AMSPORT_R0_PLC_TC3};
//  AdsDevice route{remoteIpV4, remoteNetId, 351};
  AdsVariable<double> variable_to_write{route, "MAIN.myFloatVAR"};

  ros::Rate loop_rate(0.1);

  while (ros::ok()) {
    variable_to_write = 12.0;
    ROS_INFO_STREAM("changed variable on sps");
    ros::spinOnce();
    loop_rate.sleep();
  }

  return 0;
}

```

- error : error code 6

- control pc

[image]

[image]

We are able to successfully send pings and it feels like we have tried every possible method. Is there any other solution you could suggest? Thank you.
[image]

pbruenn, 2024-04-03:
Is your PLC in RUN mode?

kkk3449, 2024-04-03:
> Is your PLC in RUN mode?

Switching to run mode and executing the code, methods 1 (C++ ADS) and 2 (Python) worked well. Method 3 (ROS) still encounters a 1797 error, but I will try modifying variables to fix it. Before editing the .xml file, it did not work even in RUN mode, but it seems to work after modifying the .xml file. Thank you.