Beckhoff/ADS #230: Add Timeout in AddRoute

martinste1n, 2024-07-04:
Hi,

first up thank you for this library! We really appreciate your efforts!

We make use of it through the [AdsToJava project](https://github.com/Beckhoff/AdsToJava). Here we face an issue of deadlocks when trying to add a route to a machine that is not available (anymore).

Our specific problem is described in an issue there:
https://github.com/Beckhoff/AdsToJava/issues/10#issue-2274717943

Would it be possible to add a timeout, so that the call will be terminated if the machine is not available?

Thanks in advance!
Martin

Relevant point in source code:
https://github.com/Beckhoff/ADS/blob/20d4a67970a6b954b32f6e51caf622da26adc13c/AdsLib/standalone/AmsRouter.cpp#L27

pbruenn, 2024-07-04:
Well, this sounds like a general TCP settings:
/proc/sys/net/ipv4/tcp_syn_retries
/proc/sys/net/ipv4/tcp_synack_retries
read: http://willbryant.net/overriding_the_default_linux_kernel_20_second_tcp_socket_connect_timeout

The problem is most firewalls are configured to silently drop  packets to closed ports or missing hosts. So your TCP stack is forced to retry. Which makes these connects appear to take forever. E.g. when I configure my pf firewall to reply with icmp on closed ports `block return in inet proto tcp from any to any port 48898` AddRoute will return immediately.

Since you mentioned a deadlock, make sure you have the latest version. We had a lock in the AddRoute facility until a few month ago: https://github.com/Beckhoff/ADS/commit/8e8fb2a9acff06cc617f351ea35dbb9e56f0cf97

martinste1n, 2024-07-04:
Thank you very much... It seems pretty promising that commit 8e8fb2a could solve our problem! Will have a look at it!