Beckhoff/ADS #232: How to Write 2D Array of Struct through ADS?

dzhydaniel, 2024-07-17:
[image]
[image]
[image]

I have a Struct and described with    [StructLayout(LayoutKind.Sequential, Pack = 1)] , and i have a  2d array type of the struct, when use ReadAny it's fine , but when i try to write the array back with WriteAny it comes back with unable to marshal issue.
Does any one know how to write 2d array with struct elements?

pbruenn, 2024-07-17:
It seems you are in the wrong repository. This one is the open source c++ standalone library. Your screenshots look like .NET. Please, contact the support for the .NET library you use

dzhydaniel, 2024-07-22:
I figured it out, just transformed 2D array to 1D array, it will work.

closed 2024-07-22 by dzhydaniel