Beckhoff/ADS #234: How to properly handle ADS error 1812: Device Notify Handle Invalid (Onlinechange, Release handle and get a new one)

xin-tsla, 2024-09-19:
Hi All, 
Wondering when encounter this error, it seems not easy to create a new one, the reason is that when receiving the handle id, there is no context info of which path in PLC we could re-create handle id for.
Is there any way to know which path for the handle it to be able to re-create?
Thanks!

soberschmidt, 2024-09-20:
Hi,
I would recommend to persist the PLC symbol on creation of the handles to cover this scenario. Furthermore I would observe the online change counter (Tc3 symbol path: TwinCAT_SystemInfoVarList._AppInfo.OnlineChangeCnt) and the symbol version (ADS Read command with IndexGroup ADSIGRP_SYM_VERSION (0xF008), Offset 0, size = 1 byte). If one of both will change, I would recommend to release all handles/notifications and create new ones.

xin-tsla, 2024-09-20:
Hi @soberschmidt , 
Thanks for the info! I will give it a try.

xin-tsla, 2024-10-25:
Hi @soberschmidt ,
Wondering when TwinCAT_SystemInfoVarList._AppInfo.OnlineChangeCnt will be changed to 0?
How twincat know the client side release all handles and reset the counter to 0?
Thanks