Beckhoff/ADS #239: The request never returns

paopaol, 2024-11-06:
When a request is sent and a response is waiting, another thread calls the delete route. Then the request will be blocked forever. After analyzing the code, it seems that the Notify of the response will never be called because the Recv method of AmsConnection will throw an exception.

```cpp
void AmsConnection::Recv()
{
    AmsTcpHeader amsTcpHeader;
    AoEHeader aoeHeader;
    for ( ; ownIp; ) {
       // Receive maybe throw expection.when socket shutdown called.
       //then the response Notify never called
        Receive(amsTcpHeader);
        if (amsTcpHeader.length() < sizeof(aoeHeader)) {
            LOG_WARN("Frame to short to be AoE");
            ReceiveJunk(amsTcpHeader.length());
            continue;
        }

        Receive(aoeHeader);
        if (aoeHeader.cmdId() == AoEHeader::DEVICE_NOTIFICATION) {
            ReceiveNotification(aoeHeader);
            continue;
        }

        auto response = GetPending(aoeHeader.invokeId(), aoeHeader.targetPort());
        if (!response) {
            LOG_WARN("No response pending");
            ReceiveJunk(aoeHeader.length());
            continue;
        }

        switch (aoeHeader.cmdId()) {
        case AoEHeader::READ_DEVICE_INFO:
        case AoEHeader::WRITE:
        case AoEHeader::READ_STATE:
        case AoEHeader::WRITE_CONTROL:
        case AoEHeader::ADD_DEVICE_NOTIFICATION:
        case AoEHeader::DEL_DEVICE_NOTIFICATION:
            ReceiveFrame<AoEResponseHeader>(response, aoeHeader.length(), aoeHeader.errorCode());
            continue;

```

pbruenn, 2024-11-20:
Where exactly? Until GetPending() the response is not marked as in use so the request will timeout. And after GetPending() I see no code path without Notify.

paopaol, 2024-11-27:
> 具体在哪里？在 GetPending() 之前，响应未标记为正在使用，因此请求将超时。在 GetPending() 之后，我看不到没有 Notify 的代码路径。

What does it mean that the response is not marked as in use? I checked the code and found that when the data is sent out, the response will wait on a condition variable. But if Receive(amsTcpHeader) throws an exception, then GetPending will never be called. Do you mean that if Receive(amsTcpHeader) throws an exception, the condition variable will also time out and be awakened by itself?

paopaol, 2024-11-28:
> Where exactly? Until GetPending() the response is not marked as in use so the request will timeout. And after GetPending() I see no code path without Notify.

What does it mean that the response is not marked as in use? I checked the code and found that when the data is sent out, the response will wait on a condition variable. But if Receive(amsTcpHeader) throws an exception, then GetPending will never be called. Do you mean that if Receive(amsTcpHeader) throws an exception, the condition variable will also time out and be awakened by itself?

pbruenn, 2024-11-28:
I repeat "Where exactly?".

Here we wait with a timeout:
https://github.com/Beckhoff/ADS/blob/20d4a67970a6b954b32f6e51caf622da26adc13c/AdsLib/standalone/AmsConnection.cpp#L25

Only after invokeId wasn't consumed by GetPending https://github.com/Beckhoff/ADS/blob/20d4a67970a6b954b32f6e51caf622da26adc13c/AdsLib/standalone/AmsConnection.cpp#L169

The Wait() will return
https://github.com/Beckhoff/ADS/blob/20d4a67970a6b954b32f6e51caf622da26adc13c/AdsLib/standalone/AmsConnection.cpp#L29

Only after invokeId was consumed we would wait a second time without timeout
https://github.com/Beckhoff/ADS/blob/20d4a67970a6b954b32f6e51caf622da26adc13c/AdsLib/standalone/AmsConnection.cpp#L33

But in these cases GetPending() was successful and we either got an unknown AMS command id
https://github.com/Beckhoff/ADS/blob/20d4a67970a6b954b32f6e51caf622da26adc13c/AdsLib/standalone/AmsConnection.cpp#L350

Or we call ReceiveFrame()
https://github.com/Beckhoff/ADS/blob/20d4a67970a6b954b32f6e51caf622da26adc13c/AdsLib/standalone/AmsConnection.cpp#L340
https://github.com/Beckhoff/ADS/blob/20d4a67970a6b954b32f6e51caf622da26adc13c/AdsLib/standalone/AmsConnection.cpp#L345

And I see no codepath in ReceiveFrame without a Notify().