Beckhoff/ADS #244: How to build the Doxygen documentation for the library?

Mebus, 2024-12-19:
Hallo,

all efforts to build the Doxygen documentation for the library ended up with a built, but empty documentation. What is the proper way to build the documentation for the library? I am not interested in the documentation for "adstool".

Thanks!

Mebus

pbruenn, 2024-12-20:
Hmm, looks like Doxygen stopped recognizing the plain C code at some point. Since I don't feel like constantly adapting to new doxygen styles, this is probably the easiest way to get an html view of the documentation:
```
diff --git a/docs/Doxyfile b/docs/Doxyfile
index 1b1ae88..5534704 100644
--- a/docs/Doxyfile
+++ b/docs/Doxyfile
@@ -398,7 +398,7 @@ LOOKUP_CACHE_SIZE      = 0
 # normally produced when WARNINGS is set to YES.
 # The default value is: NO.
 
-EXTRACT_ALL            = NO
+EXTRACT_ALL            = YES
 
 # If the EXTRACT_PRIVATE tag is set to YES all private members of a class will
 # be included in the documentation.
@@ -743,7 +743,7 @@ WARN_LOGFILE           =
 # spaces.
 # Note: If this tag is empty the current directory is searched.
 
-INPUT                  = docs/mainpage.dox AdsLib/AdsDef.h AdsLib/AdsLib.h
+INPUT                  = docs/mainpage.dox AdsLib
```

closed 2024-12-20 by pbruenn

Mebus, 2024-12-20:
Awesome! Thank you for the quick fix 😃