Beckhoff/ADS #246: Does Subcsribing non-existing ADS path cause issues?

xin-tsla, 2025-01-27:
Hi Beckhoff,

Wondering if we keep retrying to subscribe non-existing ADS pathI, (we do get ADS symbol not found error), retry every 3 seconds.
Will it cause issues? Like affecting Twincat internal ADS communications?
Looking forward to response.

Thanks,
Xin

pbruenn, 2025-01-27:
Well, of course it will affect TwinCAT. It wastes communication and processing resources. Will it hurt? Probably not, but you have to measure for yourself.

xin-tsla, 2025-01-27:
Thanks for the quick response!
The frequency of retry is one retry every 3 seconds.
Wondering does beckhoff have some way to measure it?

pbruenn, 2025-01-27:
Well, this highly depends on your usecase/application.
- Measure you applications performance.
- Then skip the notifications for non-existing ADS symbols. 
- Measure again
- compare the results

xin-tsla, 2025-01-27:
> Then skip the notifications for non-existing ADS symbols.
Wondering do you mean to not subscribe the non-existing ADS symbols?