Beckhoff/ADS #247: Is it possible to write a systemd service which replies to a broadcast search

rdkluijv, 2025-02-19:
I have the example running on an HMI panel but it doesn't show when I do a broadcast search in the router of the development PC. That would be a nice feature. The question is if it is possible using the API to write a service to respond to the broadcast search.

pbruenn, 2025-02-19:
Well, everything is possible, but this is a bit out of scope for this project. We have a client library to talk to TwinCAT routers. For this we wrote a "dummy" router, which connects to TwinCAT routers. But we have no real routing facility implemented. There is no socket listening for incoming packets/connections. So there is no UDP socket listening for the broadcast messages.
Lets say we would have that. What would be next? You would find your AdsLib client application in a broadcast search. TwinCAT would now try to add a route to itself into our router. But our router lacks this interface completely. So in the end you would have to implement a full TwinCAT router replacement.

With a bit of patience you wouldn't need to implement it. Beckhoff is working on bringing TwinCAT to Linux: https://www.beckhoff.com/en-us/products/product-news/linux-r/

rdkluijv, 2025-02-20:
Thanks for the fast respons. I was hoping for something like TC1000 TC3 ADS because that's available for BSD etc. My question is answered so the issue can be closed.