Beckhoff/ADS #250: Where can I find these macros?

jackybek, 2025-05-07:
PADSSYMBOLNAME

PADSSYMBOLTYPE

PADSSYMBOLCOMMENT

PADSNEXTSYBMBOLENTRY

I followed the sample from https://stackoverflow.com/questions/57525108/can-i-query-the-list-of-variables-and-types but get compile errors relating to the above functions.

Thanks.

Jacky

pbruenn, 2025-05-07:
They are defined in the TwinCAT SDK. This repository is a simple open source ADS client implementation distinct from the TwinCAT ADS library. If you use the TwinCAT SDK use the headers from the TwinCAT SDK. If you stick with this library from here you can take a look into https://github.com/Beckhoff/ADS/blob/e85ef696447a77039a4c3710d7573c8df32c0d58/AdsLib/SymbolAccess.h and https://github.com/Beckhoff/ADS/blob/e85ef696447a77039a4c3710d7573c8df32c0d58/AdsLib/SymbolAccess.cpp. Its our wrapper to make access to symbol information more easy.

jackybek, 2025-05-07:
Ok.  Currently i did not use the SDK from TwinCAT.  I am using your github repo to implement an ADS client.
I removed these 3 lines of lines containing these macros and now i can successfully produce the linux executable.

During runtime I get an error code 7 : target machine not found - Missing ADS routes.

I checked my code and already confirmed that a few lines above, i managed to connect to my CX9020 via ADS. My code is as follows:
int *t_ams = extractAMS(ams);
char real_ams[30];
sprintf(real_ams, "%d, %d, %d, %d, %d, %d", t_ams[0],t_ams[1], t_ams[2], t_ams[3], t_ams[4], t_ams[5]);
AmsNetId remoteNetId { real_ams };
AdsDevice route { remoteIpV4, REMOTE_NET_ID, (uint16_t)nPlcPort };
const auto state = route.GetState();
 AdsVariable<bool> readVar1 {route, "MAIN.el2024_1"};
 AdsVariable<bool> readVar2 {route, "MAIN.el2024_2"};
 AdsVariable<bool> readVar3 {route, "MAIN.el2024_3"};
 AdsVariable<bool> readVar4 {route, "MAIN.el2024_4"};

The values i received from readVar1 to readVar4 corresponded to the actual values i set in TwinCAT on my CX9020.
Next i tried the following codes:

AmsAddr Addr { { real_ams }, (uint16_t) nPlcPort };
long nErr;
long nAdsPort;
AdsVersion Version; //AdsVersion *pVersion = &Version;
char DevName[50]; //AmsAddr Addr;
//PAmsAddr pAddr = &Addr;

        char *symbols = NULL;
        struct AdsSymbolUploadInfo {
                uint32_t nSymbols;
                uint32_t nSymSize;
        } uploadInfo;

// Open communication port of the ADS router
        nAdsPort = AdsPortOpenEx();
nErr = AdsGetLocalAddressEx(nAdsPort, &Addr);

                uint32_t bytesRead;
                AdsSymbolUploadInfo tAdsSymbolUploadInfo;
                AdsSymbolEntry *adsSymbolEntry;
                Addr.netId = { real_ams };
                Addr.port = nPlcPort; // nPlcPort;
                **Addr.route = route;**

nErr = AdsSyncReadReqEx2(nAdsPort, &Addr, ADSIGRP_SYM_UPLOADINFO, 0x0, sizeof(tAdsSymbolUploadInfo), &tAdsSymbolUploadInfo, &bytesRead );

When executing the above, I get error code 7.
Since error code 7 is relating to missing ADS route, i tried to initialise the route using Addr.route = route (route was defined earlier in my code i.e. AdsDevice route { remoteIpV4, REMOTE_NET_ID, (uint16_t)nPlcPort };

But the compiler says the structure Addr does not contain member 'route'.
So how can i initialise the route created using AdsDevice and use it in AdsSyncReadReqEx2() ?

Many thanks for your help.

pbruenn, 2025-05-07:
Most of our abstractions are used in the example application "AdsTool". You find an example on how to read and write plc symbols here: https://github.com/Beckhoff/ADS/blob/master/AdsTool/main.cpp#L490

jackybek, 2025-05-07:
My intention is to get the list of variables declared in the PLC based on the port number.
For example, in my CX9020 TwinCAT 3, i have 2 PLC declared:
SolenoidValue - port 852
ProximitySensor - port 851

In PLC SolenoidValve, i declared 4 bool variables in MAIN : el2024_1, el2024_2, el2024_3, el2024_4
In PLC ProximitySensor, i have other variables declared.

So to retrieve the 4 variables in SolenoidValve, is the following functions to be called?

            auto device = bhf::ads::SymbolAccess{ gw, netid, port }; // gw = ipaddress of my CX9020, netid = AMS net of my CX9020, port = 852
            device.ShowSymbols(std::cout);

jackybek, 2025-05-07:
Successfully extracted all the info.
Thanks.

jackybek, 2025-05-07:
One more advice needed : is there any helper function that processes the ads stream? or i have to write my own ?

jackybek, 2025-05-10:
Hi Patrick,

I am facing some issues reading the value of an INT variable declared in TwinCAT3.
I have already extracted the variable symbol, but when this code is run, it says "terminate called after throwing an instance of 'AdsException'.  what() : Ads operation failed with **error code 0.**

AdsVariable<int> readVar {route, _name} ; 
where _name is GVL_Temp_probe1.

This is the source file i read from:
[image]

This is the runtime error message:
[image]

This is the declaration in TwinCAT3.
[image]

This is the output from uaExpert:
[image]

The funny thing is that if i were to pass in a literal string representing the variable name e.g. AdsVariable<bool> readVar1 {route, "MAIN.el2024_1"}, i am able to get the data.
But if i pass in a variable, i get an error code 0.
What is error code 0 ?

Thanks for your advise.

jackybek, 2025-05-10:
Resolved. It is due to an incompatible C++ variable type declared.

jackybek, 2025-05-11:
Hi,
I am trying to pipe the output of device.ShowSymbols(std::cout) to a char buffer or a file.
I tried the following code:
-----------------------------------
  std::ofstream outputFile;
  if (outputFile.is_open())
  {
        outputFile << device.ShowSymbols(std::cout) << std::endl;
        outputFile.close();
  }
------------------------------------
I only get a 2-byte file.
Can you advise if the function device.ShowSymbols() can pipe the output to a file or char buffer instead?

Thanks.

jackybek, 2025-05-13:
Hi Patrick,
Currently I am not getting any notifications whenever I change the variables in TwinCAT.

Here is the concept diagram i would like to achieve.
As of current status, i am able to accomplish the Left Hand side of the diagram. i.e. at startup steps 1-5.
However at step 3a, i need to customise the notification event by giving it additional information such as OPC nodeId.  This is to allow the notification to find a matching nodeId against the variable it receives the notification from (_name).
Based on this matching (_name -> nodeId), i will then update OPC nodes with the new value.

[image]

You can take a look at the 2 functions below.

1. NotifyCallback() - currently for testing purposes, i just want to make sure that an ADS callback is invoked whenever i force a change to the variable in TwinCAT.  But it is not responding.

[image]
2. notificationByName().
[image]

Then in my main program, for the variable i am interested to keep track, i created a notification callback : notificationByName(std::cout, route, _name, _variableType)

[image]

 i also create an array to keep every OPC node i created, along with the information i received from ADS.  This is for matching purposes i.e. to match the variable name to the OPC NodeId whenever a ADS notification is received.

[image]

At startup, notificationByName() is called for every OPCUA node created, and along with the notificationByName().

[image]
image.png

During execution, when i force change the value of a variable in TwinCAT, i did not get any response i.e. I am expecting NotifyCallback to be invoked.  But nothing happened.

I suspect there is something wrong with my code for the 2 functions i adapted from github (https://github.com/Beckhoff/ADS/blob/master/example/example.cpp).
1. NotifyCallback()
2. notificationByNameExample()