Beckhoff/ADS #252: ADS Mismatch

callen5914, 2025-05-19:
I am encountering a persistent error related to TwinCAT ADS after performing a clean installation of Windows and TwinCAT XAE 4026 (Build 4026.16). Upon launching my solution, I receive the following error:

`Could not load type 'TwinCAT.Ads.Internal.ITcAdsRaw' from assembly 'TwinCAT.Ads.Abstractions, Version=6.0.0.0, Culture=neutral, PublicKeyToken='. `

This occurs with a fresh Windows installation and fresh install and migration to build 4026 when adding an OPC UA Server to a project. Basically I followed the steps found in the Beckhoff tutorial and encountered the error which has halted the training.

MichaelKno, 2025-05-19:
This does sound like a problem with the Connectivity project or the OPC UA Server project and has most likely nothing to do with ADS, especially not with the ADS library in this repository.

Can you please send your request to support@beckhoff.com? Thank you!

callen5914, 2025-05-19:
Yes, I've submitted...

I wonder if this is normal for a fresh install. When watching the training video the error doesn't pop up...

My theory is that the OPC connectivity project is configured to look for the older 6.0.0 version of an ADS DLL. I also tried installing the newer versions via NuGet but the newest DLLs are already there. The project just asks for the older version.

RamonRodriguezCardoso, 2025-07-31:
Hi @callen5914 ,

I am experiencing exactly the same issue when trying to add an OPC UA Server. I was wondering if you were able to solve it and if so, if you can tell me how.

Thank you very much in advance!

Ramón