Beckhoff/ADS #259: Retrieving nested symbol information

pthomasj, 2025-07-31:
Is there a library function to retrieve all of the symbol information for all of the symbols in a PLC, including the nested symbols in structures and function blocks?

pthomasj, 2025-08-05:
I am getting an error:

```
/usr/local/bin$ ./AdsTool 172.16.23.46.1.1:851 plc show-symbols
2025-08-05T14:59:07-0700 Error: AdsException message: Ads operation failed with error code 6.
```

pbruenn, 2025-08-06:
Is your PLC in run?

pthomasj, 2025-08-06:
I can access the PLC with the code I wrote using the library. The issue seems to be with something that I am doing wrong when using the tool, but I'm not sure what.

pthomasj, 2025-08-13:
Thank you. I did need to add the --gw parameter. It now prints the symbol information, but it does not seem to do so for nested symbols, such as structure fields.

pthomasj, 2025-08-19:
Should I be seeing the nested symbols in structures and function blocks with the `plc show-symbols` command?

pthomasj, 2025-09-25:
@pbruenn 

This is the code that I wrote to parse the data types:

```C++
#ifndef ADS_DATA_TYPE_ENTRY_HPP
#define ADS_DATA_TYPE_ENTRY_HPP

#include "ads.hpp"

#include <cinttypes>
#include <string>
#include <vector>

namespace ads_utils {
  #pragma pack(push, 1)
  struct ads_data_type_array_info_type {
    uint32_t lBound;
    uint32_t elements;
  };
  #pragma pack(pop)

  #pragma pack(push, 1)
  struct ads_data_type_entry_header_type {
    uint32_t entry_length;
    uint32_t version;
    uint32_t hash_value;
    uint32_t type_hash_value;
    uint32_t size;
    uint32_t offset;
    uint32_t data_type;
    uint32_t flags;
    uint16_t name_length;  // excl. \0
    uint16_t type_length;  // excl. \0
    uint16_t comment_length;  // excl. \0
    uint16_t array_dim;
    uint16_t sub_items;

    void print();
  };
  #pragma pack(pop)

  struct ads_data_type_entry_type {
    struct ads_data_type_entry_header_type header;

    std::string name_string;
    std::string type_string;
    std::string comment_string;

    //std::vector<ads_data_type_array_info_type> array;
    //std::vector<std::unique_ptr<ads_data_type_entry_type>> sub_item_vec;

    void print();
  };

  std::vector<ads_data_type_entry_type> get_ads_data_type_entry_vec(long port, AmsAddr ams_address, uint32_t data_type_size);
}

#endif
```

```C++
#include "ads_data_type_entry.hpp"
#include "error_codes.hpp"

#include <cstring>

namespace ads_utils {
  void ads_data_type_entry_header_type::print() {
    printf("entry length: %" PRIu32 "\n", entry_length);
    printf("version: %" PRIu32 "\n", version);
    printf("hash value: %" PRIu32 "\n", hash_value);
    printf("type hash value: %" PRIu32 "\n", type_hash_value);
    printf("size: %" PRIu32 "\n", size);
    printf("offset: %" PRIu32 "\n", offset);
    printf("data type: %" PRIu32 "\n", data_type);
    printf("flags: %" PRIu32 "\n", flags);
    printf("name length: %" PRIu16 "\n", name_length);
    printf("type length: %" PRIu16 "\n", type_length);
    printf("comment length: %" PRIu16 "\n", comment_length);
    printf("array dim: %" PRIu16 "\n", array_dim);
    printf("sub items: %" PRIu16 "\n", sub_items);
  }

  void ads_data_type_entry_type::print() {
    header.print();

    printf("name string: %s\n", name_string.c_str());
    printf("type string: %s\n", type_string.c_str());
    printf("comment string: %s\n", comment_string.c_str());
  }

  // ----------------------------------------------------------------------------

  std::vector<ads_data_type_entry_type> parse_ads_data_type_entry_vec(std::vector<uint8_t> buffer) {
    std::vector<ads_data_type_entry_type> ads_data_type_entry_vec;

    std::size_t i = 0;
    while (i < buffer.size()) {
      ads_data_type_entry_type ads_data_type_entry;

      std::memcpy(&ads_data_type_entry.header, buffer.data() + i, sizeof(ads_data_type_entry.header));

      std::size_t name_string_start = i + sizeof(ads_data_type_entry.header);
      std::size_t type_string_start = name_string_start + ads_data_type_entry.header.name_length + 1;
      std::size_t comment_string_start = type_string_start + ads_data_type_entry.header.type_length + 1;

      ads_data_type_entry.name_string.assign(reinterpret_cast<const char*>(buffer.data() + name_string_start), ads_data_type_entry.header.name_length);
      ads_data_type_entry.type_string.assign(reinterpret_cast<const char*>(buffer.data() + type_string_start), ads_data_type_entry.header.type_length);
      ads_data_type_entry.comment_string.assign(reinterpret_cast<const char*>(buffer.data() + comment_string_start), ads_data_type_entry.header.comment_length);

      // ads_data_type_entry.print();
      // printf("\n");

      ads_data_type_entry_vec.push_back(ads_data_type_entry);

      i += ads_data_type_entry.header.entry_length;
    }

    return ads_data_type_entry_vec;
  }

  std::vector<ads_data_type_entry_type> get_ads_data_type_entry_vec(long port, AmsAddr ams_address, uint32_t data_type_size) {
    std::vector<uint8_t> read_data_vec(data_type_size);

    long error_code = AdsSyncReadReqEx2(
      port,
      &ams_address,
      ADSIGRP_SYM_DT_UPLOAD,
      0,
      read_data_vec.size(),
      read_data_vec.data(),
      nullptr
    );

    check_error_code(error_code);

    return parse_ads_data_type_entry_vec(read_data_vec);
  }
}
```

The `data_type_size` argument in the last function is provided by `ADSIGRP_SYM_UPLOADINFO2`.

This returns all of the top level data types, but none of the nested ones. For example, this is returned, but none of the sub items:

```
entry length: 216
version: 1
hash value: 0
type hash value: 0
size: 24
offset: 0
data type: 65
flags: 129
name length: 7
type length: 0
comment length: 0
array dim: 0
sub items: 2
name string: TEST_FB
type string: 
comment string: 
```

Should I be incrementing the index by `entry length` as I am doing in the `parse_ads_data_type_entry_vec` function, or is there a smaller offset that I should use to get to the sub items?

pthomasj, 2025-09-30:
Is there perhaps a picture or a detailed description of the layout in memory of the data type information available?