Beckhoff/ADS #263: Multiple processes using AdsDevice on Linux cannot connect to PLC simultaneously (connection resets / error 1861)

pere-betarobots, 2025-08-20:
## Description

We are running into an issue where two independent processes on the same Linux PC both need to connect to a Beckhoff PLC via ADS.

- When both processes try to use `AdsDevice` at the same time, only one of them succeeds.
- The other process immediately fails with `Ads operation failed with error code 1861` or later with `Broken pipe` errors.
- If only one process is running, it works perfectly.

This makes it hard to design modular ROS nodes where each node is its own process but both need PLC access.

## Environment

- OS: Ubuntu 20.04 (ROS Noetic environment)
- ADS library: 0.0.28-1
- PLC: Beckhoff TwinCAT 3 runtime (PLC running on 192.168.125.10)

## What we tried

All this tests done with the PC IP set to `192.168.125.100`

### Working scenario (one process)

- Start **one ROS node/process** using `AdsDevice` with Local AMS Net ID `192.168.125.100.1.1`, local ADS port `30000`.
- Works fine: we can read/write variables without issues.
- Also works with `.1.2` as long as it’s the **only process**.

### Also working: multiple AdsDevice in one process

- Inside one process, create **two AdsDevice objects**:
    - `192.168.125.100.1.1` → port `30000`
    - `192.168.125.100.1.1` → port `30001`
- Both coexist, read/write works.
- Suggests that the library/router is multiplexing properly inside a single process.

### Failing scenario: two processes

- Start two **independent processes**:
    - Process A → 192.168.125.100.1.1, port 30000
    - Process B → 192.168.125.100.1.2, port 30001
- Only one process connects successfully.
- The other process logs:

```
Error: read frame failed with error: Connection reset by peer
Failed to connect: Ads operation failed with error code 1861.
Error: write frame failed with error: Broken pipe
```
- Tried same AMS Net ID in both processes (`.1.1` for both), and also different AMS Net IDs (`.1.1` vs `.1.2`).
- In both cases, one process always gets dropped.

## Question

What is the recommended way to support multiple independent processes on Linux connecting to the same PLC over ADS?

- Should we expect this to work with ads-lib (and we are misconfiguring something)?
- Or is the design such that only one process per host is supported unless there’s a central ADS router service?

## Error logs (for reference)

```
[Beckhoff] Failed to connect: Ads operation failed with error code 1861.
Error: read frame failed with error: Connection reset by peer
Error: write frame failed with error: Broken pipe
[Beckhoff] Failed to connect: Ads operation failed with error code -1.
```

Would really appreciate guidance here.

Our use case: ROS nodes running in separate processes, both needing ADS connectivity. Currently this only works if we collapse everything into one process.

Thanks,

Pere

pbruenn, 2025-08-21:
You are not the first to hit this, see:
https://github.com/Beckhoff/ADS/issues/72#issuecomment-406172875
https://github.com/Beckhoff/ADS/issues/49#issuecomment-330198577
https://github.com/Beckhoff/ADS/issues/201#issuecomment-1544167471

In "standalone" mode this AdsLib does not support complex ADS client scenarios, due to the missing ADS router. However, you are lucky. TwinCAT on Linux is happening[1] and you can approach your local Beckhoff support to get access to the early access program.

[1] https://www.beckhoff.com/en-en/products/product-news/linux-r/

pere-betarobots, 2025-08-22:
The links you shared really helped clarify the situation and confirm our suspicions that the missing ADS router is the root cause here.

In the meantime, as a temporary workaround, we’re going to try using [ads-proxy](https://github.com/fengyc/ads-proxy) to provide a single ADS client endpoint that can multiplex multiple processes on our side.

Thanks for pointing us in the right direction and for maintaining this project – really appreciate the support!

tboegi, 2025-09-08:
You can try the ads-forwarder:

https://github.com/birkenfeld/ads_forwarder_rs

aatb-ch, 2025-09-27:
Hello, I just wanted to ask a last clarification regarding the TwinCAT router on linux. Installing TC including router on Linux will be possible exclusively on your new "Beckhoff RT Linux" debian-based distro, or would it be possible to have it on another distro like Ubuntu? I guess it might cause issues regarding realtime constraints, but just wondering if I will have to switch, since ROS2 provides binaries for Ubuntu it makes it quite convenient to use.. Ideally my ROS2 machine would stay on Ubuntu, so I wouldn't be able to use the native linux TC router?